asrob-uc3m/impresoras-asrob #160: Préstamo temporal de la tronxy aka Geonosis

MrTommyGuns, 2021-01-05:
Durante el segundo cuatrimestre voy a realizar un TFG donde muy probablemente tenga que imprimir piezas de tamaños superiores a los que mis impresoras de casa pueden abordar, por lo que me gustaría tomar prestada a Geonosis y llevármela a casa durante un tiempo este segundo cuatrimestre para poder realizar dichas piezas y pruebas.

Dada la situación supongo que no resultará ningún problema, por lo que si no recibo objeciones, la cogeré cuando pueda/la necesite(previo aviso por esta issue para que esté localizada y demás) . 

Trataré de hacer un perfil de cura en primer lugar y así ya de paso podemos tener algo decente para el futuro.

JorFru, 2021-01-05:
Por mí es un sí.
Además, Tommy es uno de los participantes má activos de Printer One y tiene mis bendisiones.

MrTommyGuns, 2022-02-26:
Finalmente se **devolvió** el **14/09/2021** (se me pasó documentarlo por aquí en su momento).

[image]

Fue mejorada con refuerzos estructurales, una base más estable con filamento flexible a modo de patas para que no se deslice, un cambio de ventilador de capa junto con la pieza correspondiente y luces led con un interruptor para ver como va quedando la impresión bajo unas condiciones de luz nulas, entre otras, además de actualizaciones y ajustes de marlin.

Cumplió con su objetivo y me ayudó a realizar mi TFG con honor, esfuerzo y algún que otro quebradero de cabeza.

Se pueden ver los resultados [aquí](https://github.com/MrTommyGuns/Binaural3D) y a la bicha trabajando [aquí](https://www.instagram.com/stories/highlights/17895145640163810/).

closed 2022-02-26 by MrTommyGuns