asrob-uc3m/impresoras-asrob #163: Geonosis Revival

CarlosME17, 2023-09-28:
Intento de revivir Geonosis.

## Reparar
- [ ] Mirar el estado de la mecánica. 
- [ ] Mirar el estado del firmware/Marlin.
- [ ] Mirar electrónica y conexiones.
- [ ] Probar a imprimir alguna pieza.

## Mejoras
- [ ] Reimprimir piezas mecánicas.
- [ ] Actualizar electrónica.

jgvictores, 2025-01-11:
Resumen 18 de diciembre de 2024:

- Estuvimos mirando el manual de https://github.com/bigtreetech/Manta-M8P
- Comprobamos que los jumpers estaban ya para driver UART, y hemos colocado los drivers
- Extrajimos la tarjeta SD y cambiado la contraseña del wifi (al de mi móvil, de momento)
- Accedimos a la bigtree colocado en la impresora, vía ssh

Pending https://www.klipper3d.org/ entre otras muchas cosas.

jgvictores, 2025-01-15:
De pg 31/37, la nuestra es CB1 (aparece en red como `BTT-CB1`):
Login name: biqu
Password: biqu

[Uploading BIGTREETECH MANTA M8P V2.0 User Manual.pdf…]()

jgvictores, 2025-01-15:
Note that we already have: `/dev/serial/by-id/usb-Klipper_stm32h723xx_2C002B001351313433343333-if00`

jgvictores, 2025-01-15:
The command line of `make flash FLASH_DEVICE=` was already in the bash history.

MarckArthur, 2025-01-15:
probar utilizando ortoprint usando la ip de la impresora

jgvictores, 2025-01-22:
Vía web vía IP de la impresora:
```
Klipper reports: ERROR

Option 'position_endstop' is not valid in section 'stepper_z1'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

Nótese tb que está el Klipper log ( https://www.klipper3d.org/ ) y el Moonracker log ( https://moonraker.readthedocs.io/en/latest/ ). Viendo https://www.klipper3d.org/OctoPrint.html puede interesar OctoPrint y no Moonracker, pero aún no sé lo que es nada.

jgvictores, 2025-01-22:
```bash
 $ sudo service klipper status

● klipper.service - Klipper 3D Printer Firmware SV1
     Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2025-01-15 15:38:25 UTC; 6 days ago
       Docs: https://www.klipper3d.org/
   Main PID: 801 (python)
      Tasks: 2 (limit: 1017)
     Memory: 32.7M
        CPU: 6.052s
     CGroup: /system.slice/klipper.service
             └─801 /home/biqu/klippy-env/bin/python /home/biqu/klipper/klippy/klippy.py /home/biqu/printer_data/config/printer.cfg -I >

Jan 15 15:38:25 BTT-CB1 systemd[1]: Started Klipper 3D Printer Firmware SV1.
```

jgvictores, 2025-01-22:
The files that appear via the web interface are those of `/home/biqu/printer_data/config`.

jgvictores, 2025-01-22:
> Klipper reports: ERROR
> 
> Option 'position_endstop' is not valid in section 'stepper_z1'
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

Esto según https://github.com/Klipper3d/klipper/issues/4413#issuecomment-866253490 es que no le gusta que en `generic-bigtreetech-manta-m8p-V2_0.cfg` hayan ciertas cosas en la sección `stepper_z1` (porque prefiere tirar de `stepper_z1`, los considera duplicados), así que modificada sección `stepper_z1`:

```diff
- position_endstop: 0
- position_max: 270
- position_min: -5.0
- homing_speed: 8
- second_homing_speed: 3
- homing_retract_dist: 3
+ #j#position_endstop: 0
+ #j#position_max: 270
+ #j#position_min: -5.0
+ #j#homing_speed: 8
+ #j#second_homing_speed: 3
+ #j#homing_retract_dist: 3
```

Con esto ya no salen esos errores, sino que sale toda una sección buena `System Loads`:

[image]

jgvictores, 2025-01-22:
He estado jugando con los `Endstop` y detecta bien `Triggered` (cuando no se tocan) y `Open` (cuando se pulsan) en X, Y, Z. Entiendo que Z1 no sería necesario.

[image]

jgvictores, 2025-01-22:
Desde `Dashboard` intenté un Home All, y petó:

```
Klipper reports: SHUTDOWN

Unable to read tmc uart 'stepper_x' register IFCNT
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

[image]

jgvictores, 2025-01-22:
> Unable to read tmc uart 'stepper_x' register IFCNT

Según https://www.reddit.com/r/BIGTREETECH/comments/t7fm2h/getting_unable_to_read_tmc_uart_stepper_x/ hay que """Edite para, con suerte, salvar a otra persona que obviamente no lee las instrucciones lo suficientemente bien: recibía el mensaje "No se puede leer el registro IFCNT del tmc uart 'stepper_x'" porque no había conectado la fuente de alimentación de 24 V al "Motor". "Terminales de alimentación" del Octopus, por lo que los controladores paso a paso no recibían energía. Los conecté y todo está funcionando. Me lo perdí porque no había tenido que conectar una placa base como esta antes, así que si recibe este error, verifique que sus controladores estén recibiendo energía."""

jgvictores, 2025-01-29:
`DUMP_TMC STEPPER=stepper_x`

```
17:38
Unable to read tmc uart 'stepper_x' register GCONF
17:38
========== Queried registers ==========
17:38
SGTHRS:     00000000
17:38
TPOWERDOWN: 00000014 tpowerdown=20
17:38
COOLCONF:   00000000
17:38
TCOOLTHRS:  00000000
17:38
TPWMTHRS:   00000000
17:38
IHOLD_IRUN: 00081919 ihold=25 irun=25 iholddelay=8
17:38
SLAVECONF:  00000200 senddelay=2
17:38
========== Write-only registers ==========
```

jgvictores, 2025-01-29:
https://klipper.discourse.group/t/error-message-unable-to-read-tmc-uart-stepper-x-register-ifcnt/13945/7

jgvictores, 2025-01-29:
Se ha movido!

Parece que la lógica de los end-stops está invertida, para que se moviese X al darle a home tuve que mantenerlo pulsado. ^^

jgvictores, 2025-02-05:
> PD: También sería estético/conveniente cambiar los cables que puse por unos jumpers puentecillo de toda la vida.

Hecho!

> Parece que la lógica de los end-stops está invertida, para que se moviese X al darle a home tuve que mantenerlo pulsado. ^^

WIP, see https://www.reddit.com/r/klippers/comments/zp361r/endstop_trigger_setup/

jgvictores, 2025-02-05:
> Issue now regarding movement on Z axis: one motor shakes and the other motor seems not to move. Will review similar issues, may be due to cables but also current setup.

Vale no: one Z motor shakes, the other Z motor seems not to move, but the extruder does sound.

jgvictores, 2025-02-19:
Hecho:
- run_current: 0.800 en todos
- quemar fusible al activar cama

Ver:
- https://github.com/Klipper3d/klipper/blob/master/config/printer-tronxy-x5s-2018.cfg

jgvictores, 2025-02-26:
Por documentar lo que tenemos:

- bigtreetech cb1 v2.2
- manta m8p v2.0
- tmc 2226 v1.0 (x5)

[image]

jgvictores, 2025-03-12:
Pendiente:

Ver en placa el hueco que dejó el fusible, para ubicarlo en https://github.com/bigtreetech/Manta-M8P/blob/7151bd90f7e4223c01d95c6de6a844ac7fb60438/V2.0/Hardware/BIGTREETECH%20MANTA%20M8P%20V2.0-SIZE.pdf y sacar su nombre/identificador.

En https://github.com/bigtreetech/Manta-M8P/blob/7151bd90f7e4223c01d95c6de6a844ac7fb60438/V2.0/Hardware/BIGTREETECH%20MANTA%20M8P%20V2.0-SCH.pdf se ven claramente tres fusibles (F11, F12, F13) de 15 A, pero mejor sacar cuál es por ubicación puesto que en https://github.com/bigtreetech/Manta-M8P/issues?q=is%3Aissue%20state%3Aopen%20fuse me pareció ver menciones a otros valores.

jgvictores, 2025-03-12:
La inscripción es `JF 15A`, y en foros se refieren a ellos como `1808`.

Coincide la figura exacta con [Fusible JF 15A 125V 1808 nuevo y original de JFC2410-2150FS](https://spanish.alibaba.com/product-detail/JFC2410-2150FS-new-and-original-JF-1600101002040.html).

[image]

jgvictores, 2025-03-12:
Cosas que he hecho:
1. Ajustar mejor el volumen de impresión (límites más al límite ^^) vía https://github.com/asrob-uc3m/impresoras-asrob/commit/476626dc8487dfbce9fd517ea25e8f490362faf5
2. Solicitar el fusible:

> Como reemplazo, teniendo en que lo más importante es que sea rápido, 15A y su factor forma, me parece que la mejor opción es:
> https://www.amazon.es/piezas-fusible-15A250V-rotura-JFC2410-2150FS/dp/B0B4HLQTQP
> (hay un par de opciones más: https://www.hta3d.com/es/fusible-rapido-1808-10a-15a-20a-reemplazo-para-skr-2 parece oficial pero luego no parece que tengan 15A en stock sino 10A y 20A; por otro lado https://www.amazon.es/piezas-fusible-15A250V-rotura-JFC2410-2150FS/dp/B0B4HLQTQP/ref=sr_1_1?dib=eyJ2IjoiMSJ9.vjEh4W16KrjrMOwTkpvfpQ.eR2hla-012nETyZpYAOplwQ0bw0C7ckOzr550G1dd7I también coincide en serigrafía pero pinta raruno y sería esperar por lo menos un mes).

jgvictores, 2025-03-12:
Está petando al rato con error:
> MCU 'mcu' shutdown: ADC out of range
> Sensor 'extruder' temperature -80.587 not in range 0.000:275.000
> This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.

Primero pensé que era porque quité el fusible bueno para medirlo, pero ahora leyendo el mensaje pienso que puede ser debido a la configuración del sensor (en https://github.com/asrob-uc3m/impresoras-asrob/commit/c062aca39db726131cd6737388fbd00145b2a6ff cambié el sensor, pero conste que estuvo funcionando bastante rato con esa config, ahora peta casi al momento).

jgvictores, 2025-04-03:
Fusibles!

[image]

jgvictores, 2025-04-09:
Fusible reemplazado! 🎉 

Now:
```
MCU 'mcu' shutdown: ADC out of range

Sensor 'extruder' temperature -88.203 not in range 0.000:275.000

This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

[image]

Related:

- https://www.reddit.com/r/klippers/comments/1bkc00g/why_klipper_wont_just_tell_us_which_heater_sensor

jgvictores, 2025-08-06:
Esta conexión es terriblemente sospechosa. Aparentemente:
- Sospechoso (rodeado en rojo): Cable que parece que lleva a sensor temperatura extrusor, conectado al conector de la placa identificado como VUSB que ni está marcado en manual como que se emplea demasiado.
- Tiene más sentido: Cable que parece que lleva a sensor temperatura cama, conectado al primer conector empezando por abajo de THB-TH3.

[image]

Pending también revisar tema fans.

jgvictores, 2025-08-06:
Okiii ya no hay problema de fuera de rango [0-275] (mide 28.8 grados) en extrusor!

Ahora no calienta:

[image]

jgvictores, 2025-09-24:
> * Eje extrusor.

Confirmo que intercambiando en placa conector X y extrusor, el motor extrusor se mueve como si fuese el X.